upb-lea/transistordatabase #14: Feature request: generate a virtual datasheet (pdf-file)

gituser789, 2021-05-31:
function available but not compleated
(graphs are missing)

gituser789, 2021-07-26:
Please Sort the datasheet values, make some sections:
 - Standard stuff: Manufacturer, Type, Datasheet date, Datasheet Hyperlink, 
 - Mechanical and thermal: Housing_area, Housing_type, R_th_xxxxx
 - maximum ratings (I_abs_max, V_abs_max, T_j_max, T_c_max)
 - Capacitances: C_iss_fix, ....
 - all the rest

gituser789, 2021-07-26:
Replace the 'programming variable names' by real datasheet names (clean symbols)
e.g.
C_iss_fix -> italic: 'C', subscript 'iss,fix'
Is this possible?

gituser789, 2021-07-26:
- in 'Created by', insert 'LEA' department incl. Link to Homepage (https://ei.uni-paderborn.de/lea) instead of the actual name
- Remove 'Mail ID' from the pdf

gituser789, 2021-07-26:
Just provide the english language in the virtual datasheet 
(remove 'Technische Information')

gituser789, 2021-07-27:
Check Format: A4?
graph_r_e missing

gituser789, 2021-07-27:
Add hint: 'This document is not the official datasheet
'

gituser789, 2021-07-29:
Unit of R_th_xx is K/W

gituser789, 2021-08-05:
Sorting switch conduction for temperatures is really nice!
please implement this also for the diode

Switch:
[image]

Diode:
[image]

gituser789, 2021-08-07:
Page break in case of too much graphs necessary

MohanNagella, 2021-09-21:
Fixed and updated!

closed 2021-09-21 by MohanNagella