upb-lea/transistordatabase #18: Feature Request: store diodes in the database

gituser789, 2021-06-09:


tinix84, 2022-07-11:
The switching model is only Qrr right?  Or do you plan more complex measure? 
Normally in double pulse test does not make sense to separate the diodes from the mosfet use in the double pulse, why not use a database if couple devices?

gituser789, 2022-07-11:
The switching model is Qrr and Coss, depending on the device type (PIN-Diode, Schottky, SiC-Diode). 

Yes, for DPT it does not make sense to separate diodes from switches, due to the switching behaviour is controlled by r_gate. 

Using coupled devices (switch with diode) is a good idea, but there are nearly endless of combinations out there.... 

All other parameters can be stored as usual
 * Forward characteristics
 * Cooling (r_th, housing area, ...)
 * Metadata (manufacturer, housing, ...)

So, from my point of view, the diode option is interesting to pre-select a diode for conducting losses, and may also for the Coss-losses. Qrr losses are, as you mentioned, tricky to include in the TDB.