upb-lea/transistordatabase #42: Recheck type hints

gituser789, 2021-08-06:
Wrong:
```v_abs_max: [float, int]```
first, for X OR Y, Union must be used, second, float includes int.
https://stackoverflow.com/questions/50928592/mypy-type-hint-unionfloat-int-is-there-a-number-type
Correct:
```v_abs_max: [float]```

Wrong:
```comment: [str, None]```
Better
```comment: Union[str, None]``` or ```comment: Optional[str]```
https://stackoverflow.com/questions/51710037/how-should-i-use-the-optional-type-hint

Also the list type hints are not correct.

Check: https://www.python.org/dev/peps/pep-0484/

Also in Pyhton 3.9, the typing package is in standard collection: https://www.python.org/dev/peps/pep-0585/
Introduction why the new pep: https://www.python.org/dev/peps/pep-0585/#rejected-alternatives
section about 'list' and 'List': https://www.python.org/dev/peps/pep-0585/#generics-erasure

please comment this issue for more information

MKlaedtke, 2021-08-06:
You can also add type hints to function arguments and returns like specified in PEP 484.
For example
`def isvalid_dict(dataset_dict, dict_type):`
would become
`def isvalid_dict(dataset_dict: dict, dict_type: dict) -> bool:`

gituser789, 2021-08-06:
Update first comment:
Wrong:
v_abs_max: [float, int]
first, for X OR Y, Union must be used, second, float includes int.
https://stackoverflow.com/questions/50928592/mypy-type-hint-unionfloat-int-is-there-a-number-type
Correct:
v_abs_max: float

gituser789, 2021-08-10:
Make the comment next to type-hints compatible with sphinx:
There is a ```#:``` needed as comment line
https://stackoverflow.com/questions/24080491/print-the-variable-value-in-a-sphinx-document

gituser789, 2021-08-12:
https://mypy.readthedocs.io/en/stable/cheat_sheet_py3.html

gituser789, 2021-09-06:
Find out how to give numpy-type-hints.
may this helps:    https://numpy.org/devdocs/reference/typing.html

MohanNagella, 2021-11-10:
Left with weak warnings

closed 2021-12-10 by gituser789