upb-lea/transistordatabase #61: Field to store R_ds,on vs. junction temperature

gituser789, 2021-11-15:
**Is your feature request related to a problem? Please describe.**
Provide a class where R_DS,on vs. junction temperature can be stored
[image]

**Describe the solution you'd like**
Suggestion: Provide an attribute inside switch-class, which contains a list. In this List, instances of the new class 'TemperatureDependResistance' can be stored.

Note: Sometimes the R-Value is a factor, and a nominal R-Value is given. Make it also possible so that this works.
Example: 
[image]

gituser789, 2021-11-15:
this must be checked also in the pytests (add to the test_transistor())

MohanNagella, 2021-11-15:
How can I derive actual resitance value from normalized resistance?

MohanNagella, 2021-11-15:
Also what is the intended purpose of R_ds_on vs Junc, 
Because I need to underdstand which other attributes are necessary to be added to the new class

MohanNagella, 2021-11-16:
@gituser789, is it fine to have the variable name on switch class as '**t_j_r_on**'  or do u suggest other name?

gituser789, 2021-11-16:
r_channel_th

MohanNagella, 2021-11-16:
Overview in virtual datasheet--> any comments?
[image: image.png]

MohanNagella, 2021-11-16:
solo curve
[image: image.png]

commit e3fd1964 (2021-11-16): close: #61
closed 2021-11-23 by gituser789