upb-lea/transistordatabase #66: Cannot get the corresponding datasheet according to the procedures in the tutorial

zachary121693, 2021-12-04:
Hi, I just came into contact with python and pycharm, please forgive me for asking silly questions.
When I copy the example program from the tutorial into a new project in pycharm, I get an error
show below
**To Reproduce**
K:\TDB\test2\Scripts\python.exe K:/TDB/test2/day1/T1.py
Traceback (most recent call last):
  File "K:\TDB\test2\day1\T1.py", line 2, in <module>
    import transistordatabase as tdb
  File "K:\TDB\test2\lib\site-packages\transistordatabase\__init__.py", line 1, in <module>
    from .databaseClasses import *
  File "K:\TDB\test2\lib\site-packages\transistordatabase\databaseClasses.py", line 25, in <module>
    import constants 
ModuleNotFoundError: **No module named 'constants'**

Process finished with exit code 1

Prior to this, I have successfully added the transistordatabase toolkit.

Thanks 
Zark

gituser789, 2021-12-06:
Hi, thanks for this issue.
The issue will be resolved in the next release, which will be released within the next days

zachary121693, 2021-12-06:
Hi there,

thanks for your reply.

An update:

I just manually paste the settings in constants.py to the file, which temporary  fix the problem of "no module named constants".

However, another issue happens when I try to connect the MongoDB and query the file (in tutorial "CREE_C3M0016120K").

In transistor_loaded = tdb.load('CREE_C3M0016120K'), it returns:

Note: Please make sure that you have installed the latest version of the transistor database, especially if the update_from_fileexchange()-method ends in an error. Find the lastest version here: https://pypi.org/project/transistordatabase/
Repo at K:\TDB\test2\lib\site-packages\transistordatabase\cloned_repo_TDB_File_Exchange successfully loaded.
Remote: https://github.com/upb-lea/transistordatabase_File_Exchange
switch :Thermal parameters re-assigned to foster object
Thermal parameter computation failed: graph_t_rthjc in diode's foster thermal object is empty!
Update Transistor: GaNSystems_GS66506T
switch :Thermal parameters re-assigned to foster object
diode :Thermal parameters re-assigned to foster object
Update Transistor: Fuji_2MBI100XAA120-50
switch :Thermal parameters re-assigned to foster object
diode :Thermal parameters re-assigned to foster object
Update Transistor: Fuji_2MBI200XBE120-50
switch :Thermal parameters re-assigned to foster object
diode :Thermal parameters re-assigned to foster object
Update Transistor: Fuji_2MBI300XBE120-50
switch :Thermal parameters re-assigned to foster object
diode :Thermal parameters re-assigned to foster object
Update Transistor: Infineon_FF200R12KE3
switch :Thermal parameters re-assigned to foster object
diode :Thermal parameters re-assigned to foster object
Update Transistor: Infineon_FF300R12KE3
switch :Thermal parameters re-assigned to foster object
diode :Thermal parameters re-assigned to foster object
Update Transistor: Mitsubishi_CM200DY-24T
R^2 score: 0.9994660380432201
switch :Thermal parameters re-assigned to foster object
R^2 score: 0.999551360039351
diode :Thermal parameters re-assigned to foster object
Update Transistor: Semikron_SKM400GB12T4
switch :Thermal parameters re-assigned to foster object
diode :Thermal parameters re-assigned to foster object
Update Transistor: Fuji_2MBI200XAA065-50
switch :Thermal parameters re-assigned to foster object
diode :Thermal parameters re-assigned to foster object
Update Transistor: Fuji_2MBI300XBE065-50
switch :Thermal parameters re-assigned to foster object
diode :Thermal parameters re-assigned to foster object
Update Transistor: Fuji_2MBI400U2B-060
switch :Thermal parameters re-assigned to foster object
diode :Thermal parameters re-assigned to foster object
Update Transistor: Fuji_2MBI400XBE065-50
switch :Thermal parameters re-assigned to foster object
diode :Thermal parameters re-assigned to foster object
Update Transistor: Fuji_2MBI600XEE065-50
R^2 score: 0.9988556100739272
switch :Thermal parameters re-assigned to foster object
Thermal parameter computation failed: graph_t_rthjc in diode's foster thermal object is empty!
Update Transistor: CREE_C3M0120100J
Thermal parameter computation failed: graph_t_rthjc in switch's foster thermal object is empty!
Thermal parameter computation failed: graph_t_rthjc in diode's foster thermal object is empty!
Update Transistor: CREE_C3M0016120K
R^2 score: 0.9591918844323666
switch :Thermal parameters re-assigned to foster object
Thermal parameter computation failed: graph_t_rthjc in diode's foster thermal object is empty!
Update Transistor: CREE_CAB530M12BM3
R^2 score: 0.9760705068416482
switch :Thermal parameters re-assigned to foster object
Thermal parameter computation failed: graph_t_rthjc in diode's foster thermal object is empty!
[... 23 lines cut ...]
{'_id': ObjectId('6163ea5b0f0e80a58eeaa1d2'), 'name': 'Fuji_2MBI400U2B-060'}
{'_id': ObjectId('607816ece489e3083a6e9fd0'), 'name': 'Fuji_2MBI400XBE065-50'}
{'_id': ObjectId('60781710bf5bceefc979f173'), 'name': 'Fuji_2MBI600XEE065-50'}
{'_id': ObjectId('610bbc52ddd7ddb3d78e50ec'), 'name': 'CREE_C3M0120100J'}
{'_id': ObjectId('6077f996f6c4fcaec0432799'), 'name': 'CREE_C3M0016120K'}
{'_id': ObjectId('60f0243935b6d71a76e57f1a'), 'name': 'CREE_CAB530M12BM3'}
{'_id': ObjectId('609a5fd651c1fe8a7098707b'), 'name': 'CREE_WAB300M12BM3'}
{'_id': ObjectId('610bab16c52413d164dd2884'), 'name': 'CREE_C3M0060065J'}
{'_id': ObjectId('6113bbca377f3b92d8cb313b'), 'name': 'CREE_C3M0120065J'}
{'_id': ObjectId('60a77b69fc5148b7e059f499'), 'name': 'UnitedSiC_UF3SC065007K4S'}
{'_id': ObjectId('61ae030f1cb31f7c363ca36a'), 'name': 'CREE_C3M0016120K'}
Traceback (most recent call last):
  File "K:\TDB\test2\day1\T1.py", line 11, in <module>
    transistor_loaded = tdb.load('CREE_C3M0016120K')
  File "K:\TDB\test2\lib\site-packages\transistordatabase\databaseClasses.py", line 4001, in load
    return load_from_db(collection.find_one(dict_filter))
  File "K:\TDB\test2\lib\site-packages\transistordatabase\databaseClasses.py", line 4016, in load_from_db
    if 'c_oss' in transistor_args:
TypeError: argument of type 'NoneType' is not iterable

I have no clue what is the program try to find or query, which returns a none from my local MongoDB.

Any idea on this?

Best wishes

MohanNagella, 2021-12-07:
@zachary121693 will be updated in new release

MohanNagella, 2021-12-07:
@zachary121693  will be update in new release

closed 2021-12-07 by MohanNagella

gituser789, 2021-12-09:
version 0.3.0 released, please update
pcharm: File -> Settings -> Project: YourProjectName -> Python Interpreter. Chose transistordatabase, and press the ^ button top left

commit 728e0410 (2021-12-07): close #66: @zachary121693 will be updated in the new release