upb-lea/transistordatabase #70: TDB File Exchange: Json files are stored in one line

wkirgsn, 2021-12-10:
**Describe the bug**
json files consist of one single line.
This is difficult to version and small changes get impossible to detect since the whole line is marked as changed.

**To Reproduce**
Look into json files.

**Expected behavior**
Json files could be stored with carriage returns after each property.
small changes would be local to the property line.

**Screenshots**
[image]

**Desktop (please complete the following information):**
- All OS

**Additional context**
Visible in github already

MohanNagella, 2021-12-10:
@wkirgsn the issue is fixed and is pushed to main branch. Please pull in the new changes

closed 2021-12-10 by MohanNagella

wkirgsn, 2021-12-10:
The Fileexchange seems unaltered

MohanNagella, 2021-12-10:
the current fix will generate the required indentation (i.e to display in object format) while exporting the transistor object to json file format using export_json function of transistordatabase.--> this is how we extracting json files

The json files that already present in the file_exchange repository doesn't include the current fix and hence we need to update the entire file_exchange once again.

If it is one file that you want to edit or view then please load the corresponding transistor and using updated export_json method. Therefore you can obtain a pretty json file format now. We will update the entire file exchange by tommorrow.

gituser789, 2021-12-14:
Transistors in Fileexchange are updated

commit 81d8805a (2021-12-10): close #70: added indentation in json exporter