upb-lea/transistordatabase #74: GeckoCircuits temperature interpolation not working for all transistors from TDB

gituser789, 2022-01-10:
**Describe the bug**
Some termperature interpolations inside geckocircuits are not working for all transistors.
Currently is is not clear if there is a bug in geckocircuits or inside the TDB.

**To Reproduce**
Steps to reproduce the behavior:
1. load transistor CREE_C3M0065100J 
2. export it to geckoCircuits with r_g = 3 Ohms (may also default r_g works)
3. load data into geckocircuits
4. Interpolate the channel for other temperatures (90°C in the screenshot example) - this is not working
5. Find out if the problem is within geckocircuits or the data export from the TDB
6. The same procedure works for CREE_C3M0060065J!
7. See Screenshots!

**Expected behavior**
Temperature interpolation works fine

**Desktop (please complete the following information):**
Current development version of TDB

[image]
[image]

MohanNagella, 2022-01-17:
Possible issue with Gecko Circuits as no issue found with the code lines #1678-1681 which copy the channel curves from the existing database

MohanNagella, 2022-01-17:
[image]

closed 2022-01-17 by MohanNagella