upb-lea/transistordatabase #80: What is the purpose of generating models for simulation software?

jimmy-zhangjinshui, 2022-03-10:
HI Nikolas, what is the main purpose of generating models for software? To evaluate the transistors' thermal performance or dynamics? Because it seems that different software accept different levels of model, e.g. PLECS won't support SPICE model. How do we handle this problem?

gituser789, 2022-03-11:
Hi Jimmy,

in most programs, you can load an existing transistor models (e.g. provided by the manufacturer) or generate your own transistor model using datasheet values (time-consumpting). With these models, you can get the power losses or the junction temperature and so on from your circuit simulator. So you can run an optimization, watch dynamic behaviour, or whatever you are interested in.

So the main benefit is to save time when creating a transistor model for an external circuit simulator (if it is already available in the [transistor database file exchange](https://github.com/upb-lea/transistordatabase_File_Exchange)).

Yes, the exporters are very different, depending on the needs of the external circuit simulator. So there is a need for every external program to have a separate exporter. By the way, i think we can not support SPICE simulators (needs time-domain behavior: voltage and current curves during switching process and calculates the losses out of it), because the structure is very different from plecs/geckoCIRCUITS/Simulink (working with look-up-tables, what we can provide).

Best regards,

jimmy-zhangjinshui, 2022-03-16:
Thanks, that makes a lot sense!

closed 2022-04-12 by gituser789