upb-lea/transistordatabase #89: [Feature request] Find out differences in local database vs. TDB_FE database

gituser789, 2022-06-23:
**Is your feature request related to a problem? Please describe.**
To get an overview what transistor is locally different to the tdb file exchange (TDB-FE), it would be nice to have a function comparing these transistors.

**Describe the solution you'd like**
 - Button to compare transistors (local database vs. TDB-FE)
 - differences should be marked 
 - details need to be shown: where is the difference exactly?

gituser789, 2022-06-23:
Note:
While working with different transistor versions, the automatic 'update_from_fileexchange' at program startup should be removed. Update only manually by the already existing button.

JonasHoelscher, 2023-03-21:
Added in e1f2fdec0e229a2241128a81daf677709c2aec4d.

closed 2023-03-21 by JonasHoelscher