upb-lea/transistordatabase #91: Find follow-up way to generate pip-packages

gituser789, 2022-06-23:
**Is your feature request related to a problem? Please describe.**
[image]

**Describe the solution you'd like**
Find follow-up way to generate pip-packages

JonasHoelscher, 2023-03-21:
It seems setuptools with a setup.py is deprecated. There are now 2 possibilities:
1. Hotfix:
You can get rid of the warning using an older version of setuptools. The following worked for me:
`python.exe -m pip install --upgrade pip
pip install --upgrade pip setuptools wheel`
-> This updates it to the newest version which then throws a similar exception:
> SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
Then it can be downgraded to 58.2.0:
`pip install setuptools==58.2.0`
Maybe you can diretly downgrade it to 58.2.0, but  I haven't tested this on my system.

2. Use the new format
It seems that from now on the setup.py is not longer needed. There is a different file called pyproject.toml which must be created. More information in the following links:
- https://packaging.python.org/en/latest/tutorials/packaging-projects/
- https://setuptools.pypa.io/en/latest/userguide/pyproject_config.html
- https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html#summary

gituser789, 2025-05-21:
fixed by 3f57b3a1951507063fd180d69f639e47a601f851

closed 2025-05-21 by gituser789